UCI-HyperXite/hx9-control-system #26: [GUI] Connect control panel buttons to emit events for pod actions

taesungh, 2024-03-13:
In conjunction with #21, the pod action buttons in the control panel will need to be able to control the pod
- Add methods to emit events for pod actions (e.g. `sendService`, `sendStart`, `sendStop`) to `PodSocketClient`
  - Refer to last year's [`PodSocketClient.ts` from UCI-HyperXite/hx8-control-system](https://github.com/UCI-HyperXite/hx8-control-system/blob/main/control-station/src/services/PodSocketClient.ts)
- Add `onClick` handlers to the control panel buttons to invoke the corresponding method to send pod actions

ryescholin, 2024-05-13:
Hey @taesungh i think this got included in #19, are we good to close this issue?

taesungh, 2024-05-13:
Yes, we did connect the emitters, but we can clean up some parts we don't need
- Since we're using acknowledgements, we don't need to specify the additional events in `ServerToClientEvents`
- For the pod actions, the message already indicates the type, so no additional data is needed: we can update the corresponding types in `ClientToServerEvents` to `(ack: (data: string) => void) => void`
  - Accordingly, we can remove the `Data(data)` extractor and field in the corresponding `handle_<action>` functions in `StateMachine`

ryescholin, 2024-05-13:
gotcha!