UCI-HyperXite/hx9-control-system #27: [Pod] Implement current measurements from INA219 over I2C

taesungh, 2024-04-02:
For the braking system, we'll need to measure the pressure transducers to monitor the upstream and downstream pressure of the pneumatic system. These pressure transducers emit a variable current which we measure using an [INA219](https://www.adafruit.com/product/904) which communicates with the Raspberry Pi via I2C.
- Create an abstraction for the pressure transducers as a `struct`
  - Maintain and clean up an [I2c instance from RPPAL](https://docs.golemparts.com/rppal/latest/i2c/struct.I2c.html)
- Provide a method to get the pressure reading based on the current reading from the INA219
  - We can assume a linear scaling factor for now
- We can handle integration with the FSM separately

We will also eventually want to incorporate a thermistor with a similar current measurement system, so it may help to provide an intermediate abstraction for using the INA219, but that will depend on the complexity of the I2C communication.

taesungh, 2024-04-08:
Update: we will also need a way to parse the data packets provided by the INA219. We could try doing this ourselves or look into using a library such as [scttnlsn/ina219](https://github.com/scttnlsn/ina219). This library seems to be based on `embedded_hal`, and RPPAL should implement the appropriate traits for the I2C peripheral to be compatible.

samderanova, 2024-04-19:
Worked with @c11whaley this past week to set up a simple INA219 circuit with a potentiometer and tried using the Rust example provided in the [scttnlsn/ina219](https://github.com/scttnlsn/ina219) repository. However, the script throws an input output error. This doesn't seem to be an issue with the circuit because the INA219 is showing a green light.

taesungh, 2024-04-23:
As mentioned in subteam meeting, perhaps your program was using a different I2C address than the INA219 you're testing on? You can run `i2cdetect -y 1` on the Raspberry Pi to list the available I2C addresses.

closed 2024-05-03 by samderanova