UCI-HyperXite/hx9-control-system #56: [GUI] Display state transition messages in `Console`

taesungh, 2024-05-18:
Similar to #50, we can display messages in the `Console` component with status updates from the pod. For now, we can start with messages for state transitions as they happen.
- Create a new `Message` interface in `PodSocketClient` to represent a console message with a `timestamp` and `message` property
- Add a new `messages` array property to the `PodData` interface
- In the acknowledgement callback of the action messages, push a new message indicating the state has changed
  - We'll eventually also want a more generic status message handler for other notices from the pod
- In the `Console` component, display the array of console messages from `podData`

taesungh, 2024-05-30:
@vrushang1234, any progress on this?

taesungh, 2024-05-31:
@ryescholin, you wanna have a shot a this instead then?

vrushang1234, 2024-05-31:
> @vrushang1234, any progress on this?

I have been struggling to make a context for podData

samderanova, 2024-05-31:
> > @vrushang1234, any progress on this?
> 
> 
> 
> I have been struggling to make a context for podData

We actually already merged a context into `main`. See [PodContext.ts](https://github.com/UCI-HyperXite/hx9-control-system/blob/main/control-station/src/services/PodContext.ts).

PR #107 (2024-06-01): Display state transition messages in `Console`
Store `messages` in `PodData` to display in `Console`.
Resolves #56.
closed 2024-06-01 by vrushang1234