UCI-HyperXite/hx9-control-system #57: [Pod] Provide abstraction for high-voltage system

taesungh, 2024-05-18:
Similar to #28, to operate the inverter for the LIM, we have a port on the relay hat to trigger the [contactors](https://en.wikipedia.org/wiki/Contactor) for the high-voltage system (HVS). We'll need an abstraction to enable and disable the contactors through the relay's corresponding GPIO pin. This should be nearly identical to how we operate our `SignalLight`.
- Create `HighVoltageSystem` struct to operate the relay's GPIO pin via `enable` and `disable` methods
- We will eventually need more logic in the FSM for interacting with the HVS, but that can be handled separately

ryescholin, 2024-05-24:
hey what pin # is this @taesungh

taesungh, 2024-05-24:
From [last year's `high_voltage_system.py:8`](https://github.com/UCI-HyperXite/hx8-control-system/blob/main/pod-control/src/components/high_voltage_system.py#L8)
```python
PIN_CONTACTOR_RELAY: Final[int] = 20
```